SocketMobile/swift-package-capturesdk #1: Setup scan symbologies from CaptureHelper?

chbeer, 2022-08-15:
[image]

Can we somehow enable and disable scan symbologies from CaptureHelper? We need the DuraScan 740 to ignore 1D Barcode and just scan QR-Codes.

socketPM, 2022-08-15:
@chbeer yes, you can refer to the documentation here:https://docs.socketmobile.com/capture/ios/en/latest/topicsActivityLevel.html#enabling-or-disabling-a-barcode-symbology
We don't have a way to identify 1d or 2d. So you will have to disable all the 1d types individually and only enable QR Code.

cyrille-socket, 2025-04-10:
Closed as there's no activity.

closed 2025-04-10 by cyrille-socket