SocketMobile/swift-package-capturesdk #2: no such module 'CaptureSDK'

tomcondon, 2022-08-25:
I am trying to move from your CocoaPod based library to the new Swift Package Manager library and am running into the following issue using Xcode 13.4.1 (same issue on the latest beta of Xcode 14). When compiling, I get

no such module 'CaptureSDK'

This is really odd since Xcode gladly completes the import of CaptureSDK but the compiler cannot find it. Any Ideas?

-Tom Condon

cyrille-socket, 2022-08-25:
Hi Tom,

can you tell me which version of the package you're using?

Did you try to clean the DerivedData? The latest Xcode release is really prone to errors on resetting the SPM caches properly and taking the latest versions.

My hint is: in the "Package Dependencies" section of the project explorer in Xcode to right click and
1. Reset caches
2. Update package

If the error still appears, then I'm gonna need more details on your project. What kind of project?  Swift only? ObjC/Swift?

Cyrille

tomcondon, 2022-08-25:
I am using what SPM pulls down. Had an error the first time I added the CaptureSDK package. Removed it, and added again it worked. I am very familiar with the need to Reset Caches and Update packages, I have to do that all the time for my other Swift packages. Ugh.

Using:
CaptureSDK Version 1.5.2

So here is a little more info. I made a wild guess and went to the Add Binaries with Libraries section of build phases. Sure enough, no CaptureSDK. I added it and no longer have the compilation issue. But sure enough, another issue arose. The build reported an error:

Showing Recent Messages
'/Users/xxxxxx/Library/Developer/Xcode/DerivedData/EyeRepSwift-gyfgswcitdtwdcbefijhmxchvdmd/Build/Products/Debug-iphoneos/CaptureSDK.framework/CaptureSDK' does not contain bitcode. You must rebuild it with bitcode enabled (Xcode setting ENABLE_BITCODE), obtain an updated library from the vendor, or disable bitcode for this target. file '/Users/tcondon/Library/Developer/Xcode/DerivedData/EyeRepSwift-gyfgswcitdtwdcbefijhmxchvdmd/Build/Products/Debug-iphoneos/CaptureSDK.framework/CaptureSDK' for architecture arm64

So was this intended for CaptureSDK to be built with bitcode off? Anyway, the enable bitcode setting in build settings seems no longer to be there so I am trying  to figure out where to go from here.

tomcondon, 2022-08-25:
OK, a little more information. The stupid "type" menu in Build Settings was not set to all so it would not find the bitcode setting. I changed it to all and found the enable bitcode setting, set it to NO and did a full  clean build of my project. That works. No more bitcode I guess.

cyrille-socket, 2022-08-26:
Thanks @tomcondon for the details. Indeed we will update the doc with this Bitcode info.
I think that having Bitcode enabled blocks our SDK to compile but I'll try again to see if we can change that.

closed 2022-08-26 by cyrille-socket