SocketMobile/swift-package-capturesdk #4: Test Flight downloaded app does not seem to work with the scanner

tomcondon, 2022-09-07:
This is a very odd one. I updated to the latest library (Capture version 1.5.2.676) and my scanning code works fine when I run it from Xcode. However, when I uploaded it to TestFlight and then downloaded it to my device with a properly paired S700 scanner, the scanner code seems to fail silently.

Any ideas of what is happening?

tomcondon, 2022-09-08:
FYI: converting back to the latest CocoaPods version of this library seems to work fine. I think you have issues with your SPM library.

cyrille-socket, 2022-09-08:
Hi @tomcondon . Do you have any error code or something?
Does the scanner connect to your app?
Does the scanner beeps?
Do you manage to scan barcodes?
Also, the latest version is 1.5.5 if you can try. We have enabled Bitcode.

cyrille-socket, 2024-07-15:
No more activity on this thread since more than 1 year.

closed 2024-07-15 by cyrille-socket