SocketMobile/swift-package-capturesdk #5: Extra null character

ian-sayles, 2022-09-09:
After changing from the pod sdk to the swift package when getting the RFID number an extra null character is present at the end of string, so getting 17 characters rather than the expected 16.

At the moment I have added the following to my app to remove the extra value, which fixes the issue:
let string = decodedData?.stringFromDecodedData()!.replacingOccurrences(of: "\0", with: "")

Was it intended for the null value to be added or is it just a bug?

cyrille-socket, 2022-09-09:
Hi @ian-sayles. 

What was your pod version?
What is your SPM version?
What is the device you're using?

Then, can give us some sample code to know what you're doing exactly please?

It's hard to say if it's a bug given the info you provide.

Thank you
Cyrille

ian-sayles, 2022-09-12:
Hi Cyrille,

SPM Version 1.5.5
iPad Pro 2nd Gen,  iPadOS 15.6.1
D600

Below is the function:
func didReceiveDecodedData(_ decodedData: SKTCaptureDecodedData?, fromDevice device: CaptureHelperDevice, withResult result: SKTResult) {
            
            if result == .E_NOERROR {
                    
                // cancel time out for scanner
                self.dispatchWorkItem!.cancel()
                
                if self.searching {
                    if decodedData != nil {
                        let string = decodedData?.stringFromDecodedData()!.replacingOccurrences(of: "\0", with: "")
                        self.searching = false
                        
                        if decodedData!.dataSourceName!.starts(with: "ISO 14443") {
                            self.data = reverseTagId(tagValue: String(string!).uppercased())
                         } else {
                            self.data = String(string!).uppercased()
                             
                             
                        }
                    }
                }
                captureHelper.popDelegate(self)
             }
        }

The POD version that worked:

SKTCapture (1.3.133)

Regards
Ian

cyrille-socket, 2022-09-14:
Hi @ian-sayles,

thank you for the info. We'll get back to you after an analysis.

Best,
Cyrille

closed 2023-07-13 by ian-sayles