SocketMobile/swift-package-capturesdk #6: Slices for x86_64

mortenbekditlevsen, 2022-09-29:
Hi there,
Thank you so much for adding SPM support. This makes the library much easier to work with.

Would it be possible to also add a (non-simulator) x86_64 slice?
Our iOS app runs on iPhone, iPad and macOS using M1 or M2 chips and we would really like to support intel Macs as well through Catalyst.

cyrille-socket, 2022-09-29:
Hi @mortenbekditlevsen,

we're planning to do a SDK for macOS but it shouldn't support the "classic" Bluetooth devices to start but BLE for sure. What kind of devices you're using?

Do you have an email so we can send you an "alpha" version for you to test?

Thank you
Cyrille

mortenbekditlevsen, 2022-09-29:
Our customers are currently using classic bluetooth devices, the 7Ci, 7Xi and similar.
Is there a BTLE-alternative to that form factor of devices? That would certainly be interesting. 

I would love to receive an alpha version on "morten at ka-ching dot dk" :-)

Sincerely,
/morten

cyrille-socket, 2023-02-10:
Hi @mortenbekditlevsen 

we are preparing a release for a new device next week, and then we'll try to make the framework available for Mac Catalyst with the architecture x86_64 slice, but it seems not that easy regarding our project configuration.

I'll keep you posted.

Cyrille

cyrille-socket, 2026-02-04:
Closing. No activity for a long time.
We plan to make a macOS CaptureSDK flavor later this year. It will support BLE first.

closed 2026-02-04 by cyrille-socket