SocketMobile/swift-package-capturesdk #7: String extension conflicts

a-Mo7, 2022-10-05:
Hello !  

Today, I tried to switch from pod to Swift package but I encountered multiple conflicts with a String extension present in both your package and [CryptoSwift](https://github.com/krzyzanowskim/CryptoSwift) one.  

The problem is, even without importing CaptureSDK in my Swift file, I am having error like this where I import CryptoSwift:  
`Ambiguous use of 'sha256()'`   

Indeed, both packages have a String extension function named 'sha256'.

I looks like the extension present into your package is visible everywhere in the package target, even without importing **CaptureSDK** on top of the file.  

However this is not the case for the CryptoSwift String extension, because conflicts disappear when removing the import.

Please, could you keep your extension functions isolated ? As well as Types, enums, etc.

cyrille-socket, 2022-10-06:
Hi @a-Mo7,

we'll look into it. Good catch.

Cyrille

socketPM, 2022-10-10:
@a-Mo7 we are looking into this. Will get back to you on that asap. Thanks.

cyrille-socket, 2023-02-17:
@a-Mo7 The issue has been addressed with the latest version of the SDK released yesterday: https://github.com/SocketMobile/swift-package-capturesdk/releases/tag/1.6.30

closed 2023-02-17 by cyrille-socket