SocketMobile/swift-package-capturesdk #9: Swift Package: CaptureSDK framework is missing in one of my targets, in a two target project.

jrscorca, 2023-03-07:
When I add CaptureSDK via Swift Package manager, the SDK is integrated fine for the target that I select.  On my 2nd Target, CaptureSDK is not found and there is no way to add it.  A runtime error occurs:

`Library not loaded: @rpath/CaptureSDK.framework/CaptureSDK`

You can look at Build phases -> Link Binary with Libraries and see that one target has CaptureSDK, and the other target does not.  pressing "+" and trying to add manually, CaptureSDK does not appear.

Is there a workaround for this?

cyrille-socket, 2023-03-07:
Hi @jrscorca 

it seems like a Xcode issue regarding SPM.

When searching on the web, this post (https://stackoverflow.com/questions/57625952/consume-swift-package-for-multiple-targets-and-platforms-in-a-project) is often referred but those solutions don't work too and other answers are outdated.

I suggest you file a bug report to Apple as it is not our SDK and many people seem to have the same issue than you.

Best,
Cyrille

cyrille-socket, 2023-03-07:
HI @jrscorca 

I know it's not an ideal solution but if you use CaptureSDK with Cocoapods it works as you can add a pod to any target in z Xcode project.

I agree that now the future is Swift Package Manager and Cocoapods is almost something of the past but waiting for Apple to improve Xcode....

Best
Cyrille

jrscorca, 2023-03-09:
Hi @cyrille-socket , I found this issue by migrating away from Cocoapods.  Oddly, only the Socket Mobile package has this problem.  I migrated Firebase to Swift Package Manager as well and I don't have any problems with the target.  I'm unfamiliar with how the process works...but are there any configuration settings in your control that could cause the framework to be excluded in multiple targets?  

If I remove the captureSDK package entirely and build for the Target B, Target A won't have the captureSDK framework either.  Strange that Firebase doesn't seem to have this issue.

cyrille-socket, 2023-05-02:
Hi @jrscorca 

the issue is fixed with Xcode 14.3. I have been able to add CaptureSDK as a Swift Package Manager in two targets in a project.

You just have to link the binary in Build Phases manually in the second target.

Best,
Cyrille

closed 2023-05-03 by cyrille-socket