SocketMobile/swift-package-capturesdk #12: D600 app crash

ian-sayles, 2023-07-13:
My app works fine with the D600 when using a single D600, and there are no other ones around.

But if there is another D600 switched on but not connected to another iPad it causes my app to crash with the following:

Thread 11: EXC_BAD_ACCESS (code=1, address=0x0) in ibsystem_platform.dylib`:

I have stripped the code back to bare minimum to see if I can fix the issue but it still causes the crash.

To reproduce the issue.

Switch on two D600, one that is paired to the iPad and another that is not.
Start my app and when start reading RFID  app connects to the D600 which is paired to the iPad and works ok, the message to pair with the other D600 appears on the iPad, click no and continue to read tags, after a few reads the above errors occurs.

socketPM, 2023-07-13:
@ian-sayles With D600, you don’t need to pair it in the iOS Settings in advance. As long as you make sure the device is in the right mode (reader mode?), your app should be able to connect to it through Capture’s device manager. Also, are you using just one * for you favorite (device)? You can actually specify the Bluetooth ID for the favorite. That way your app will tries to fin connect to the device and stop searching. And what is the firmware version of your D600? More traces will be more helpful too. 
Thanks

ian-sayles, 2023-07-17:
Thanks for the suggestion about using favourite, have changed my app to use this, and it now works fine, just took me a while to understand how it works.

Thanks