SocketMobile/swift-package-capturesdk #13: Disabling SocketCam C820 still calls didNotifyArrivalForDevice

chbeer, 2023-10-26:
I disabled the SocketCam C820 as described here:
https://docs.socketmobile.com/capture/ios/en/latest/socketCam.html#enabling-socketcam

I just call it with `.disable`, but I get `E_INVALIDHANDLE`, then the SocketCam still reported with `didNotifyArrivalForDevice`

cyrille-socket, 2023-10-26:
Hi @chbeer,

ok I'll look into it.

In the meantime you can filter out the SocketCam device that arrives with

`if deviceInfo.deviceType == .socketCamC820 {
      return
}
`

Best,
Cyrille

yliu342, 2023-12-12:
Hi,
Any update on this? We are also facing the same issue. We didn't setup SocketCam and it seems to be automatically added and received with the `didNotifyArrivalForDevice` method.

It only happens on the latest 1.8.34

yliu342, 2023-12-12:
When we call the `getDevices()` the `. socketCamC820` is always returned. We will have to manually filter it out..

cyrille-socket, 2023-12-14:
Hi @yliu342 

the version 1.8.51 on our private git repository also fixes this issue of having SocketCam device in your list.

yliu342, 2024-01-08:
Hi @cyrille-socket 

I'm using the latest 1.8.53 there is a new entry you need to add to `LSApplicationQueriesSchemes` which is not mentioned anywhere in the doc.

Also the C820 is still showing if there are no other socket scanner is connect. If there is a connected socket scanner then C820 is not showing.

cyrille-socket, 2024-01-08:
Hi @yliu342 

If you're using the 1.8.53, it means that you got the CaptureSDK on our private repository. Indeed, it has been removed from the Readme. It will be reinserted in the next release. Here's the part:
```
You also need to add the following key to your Info.plist: LSApplicationQueriesSchemes (Queried URL Schemes) with a new item: sktcompanion (in lower case).
```

So as of the 1.8.53, CaptureSDK enables by default SocketCam so if you don't have a physical scanner, you still can scan a barcode out of the box.
So normally if you have a physical scanner connected, you should have it still unless you disabled it.

Can you share the code of your `didNotifyArrivalForDevice` please?

yliu342, 2024-01-08:
Hi @cyrille-socket we don't do anything special in `didNotifyArrivalForDevice` basically every time `didNotifyArrivalForDevice` get called we make a call to `getDevices()` and update a list of connected devices. The C820 is always returned from the the `getDevices()` function.

I also tried to set before calling `openWithAppInfo`
```
socketAPI.setSocketCamStatus(.disable) { result in
            print("\(result)")
        }
```
And the result is always `E_INVALIDHANDLE`

yliu342, 2024-01-08:
Also since your company moved to the private repo in future how to we get support or open issues such like this?

cyrille-socket, 2024-01-09:
> Hi @cyrille-socket we don't do anything special in `didNotifyArrivalForDevice` basically every time `didNotifyArrivalForDevice` get called we make a call to `getDevices()` and update a list of [...]

`openWithAppInfo` needs to be called in the first place along when you launch your app basically. Then ou can call any set/get property.

cyrille-socket, 2024-01-09:
> Also since your company moved to the private repo in future how should we get support or open issues such like this?

You can write to our developers support email: `developers@socketmobile.com`

cyrille-socket, 2024-11-20:
Hi @Ceroce , the device arrival of SocketCam C820 is normal as it is enabled by default in our CaptureSDK. If you don't use camera scanning, you can do the following:
```swift
if deviceInfo.deviceType == .socketCamC820 || deviceInfo.deviceType == .socketCamC860 { 
        // do something or don't do something 
}
```

or disable SocketCam with our `CaptureHelper`:

```swift
let captureHelper = CaptureHelper.sharedInstance
captureHelper.setSocketCamStatus(.disable, withCompletionHandler: { (setResult) in
        if setResult == SKTCaptureErrors.E_NOERROR {
                print("SocketCam disabled returned \(setResult.rawValue)")
        }
})
```