SocketMobile/swift-package-capturesdk #15: New SDK from your portal signature can't be verified by Xcode 15.1

yliu342, 2023-12-12:
Hi @cyrille-socket I noticed you update the repo in the past day and when trying to integrate the new SDK 1.8.47 using the OAuth link from your developer portal in Xcode 15.1 with SPM I get error:
```
The signature of "CaptureSDK.xcframeowrk" cannot be verified.
A sealed resource is missing or invalid.
```

cyrille-socket, 2023-12-13:
Hi @yliu342,

yes I can reproduce it indeed. I'm gonna try to find a fix. In the meantime, you can use the deprecated version 1.8.34 on Github.

cyrille-socket, 2023-12-14:
Hi @yliu342 ,

I managed to fix the signature validation issue on the SPM. So the new version is 1.8.51.

Please let me know if that works for you.

Cyrille

yliu342, 2023-12-18:
Tried 1.8.51 seems to be working

yliu342, 2023-12-18:
Something maybe worth checking. It failed to resolve package on Xcode 14.2 but when I updated to Xcode 15.1 it works fine.
The error from Xcode 14.2 is listed below
```
[01:52:11]: ▸ Loading...
[01:53:53]: ▸ xcodebuild: error: Could not resolve package dependencies:
[01:53:53]: ▸   local binary target 'CaptureSDKBinaryTarget' does not contain expected binary artifact named 'CaptureSDKBinaryTarget'
[01:53:53]: ▸   fatalError
xcodebuild: error: Could not resolve package dependencies:
  local binary target 'CaptureSDKBinaryTarget' does not contain expected binary artifact named 'CaptureSDKBinaryTarget'
  fatalError
[01:53:53]: Exit status: 74
```

cyrille-socket, 2023-12-18:
Thank you for your feedback.

For the failure of resolving the package, sometimes Xcode does not really handle well the cache of the SPMs so the cache folder needs to be deleted.

closed 2023-12-18 by cyrille-socket