SocketMobile/swift-package-capturesdk #19: Incorrect privacy manifest

yliu342, 2024-04-12:
@cyrille-socket When submit app for review we get warnings regarding `ITMS-91053: Missing API declaration`. After looking at the app bundle, I saw `Capture SDK` and `SwiftDecoder` are the one that uses those required API.

Using version Capture SDK 1.8.57 and SwiftDecoder 5.8.14 I can see both SDK are statically linked to our app.
Capture SDK has a PrivacyInfo.xcprivacy file within the CaptureSDK.framework. But this file doesn't declare any use of those required api meaning the `NSFileCreationDate`.
SwiftDecoder SDK doesn't have a PrivacyInfo.xcprivacy file at all.

Can we address this with a release or at least provide some guidance what the SDK uses those API for?
NSPrivacyAccessedAPICategoryFileTimestamp
NSPrivacyAccessedAPICategoryDiskSpace
NSPrivacyAccessedAPICategorySystemBootTime

```
SwiftDecoder.framework/SwiftDecoder: fstat, mach_absolute_time, NSUserDefaults, stat
CaptureSDK.framework/CaptureSDK: NSFileCreationDate
```

ios-dev-socket, 2024-05-16:
Hi @yliu342 . We just released the version `1.9.30` of our SDK on our Developer Portal which contains the Privacy Policy for SwiftDecoder.

cyrille-socket, 2024-07-15:
This issue has been resolved with version `1.9.30`.

closed 2024-07-15 by cyrille-socket