SocketMobile/swift-package-capturesdk #21: Cannot disconnect socket mobile S700

huynhlqDev, 2024-11-13:
I am implementing app functionality with connecting multiple peripherals and some case need disconnect socketMobile device.
I tried to use: (1)
`captureHelper?.captureApi?.close() { result in`
`            if result == SKTResult.E_NOERROR {`
`               print("captureApi closed")`
`            } else {`
`                print("can not close captureApi: \(result)")`
`            }`
`        }`
or  (2)
`captureHelper?.closeWithCompletionHandler() { result in`
`           if result == SKTResult.E_NOERROR {`
`               print("captureHelper closed")`
`            } else {`
`                print("can not close captureHelper: \(result)")`
`            }`
`        }`

but it doesn't work. In addition, running both causes the app to crash (related to BLE).
Can only turn off mobile socket with physical button [off]. but don't know what is the cause

EricGlaenzer, 2024-11-13:
Closing capture does not disconnect the scanner. In most cases it is not appropriate to disconnect the scanner from the host as it could cause a lot of confusion for the end user. The best is to power off the scanner. There is a command to power off the scanner that you could use but again often this causes a not so great user experience. We will look at this BLE crash, although make sure you are using the latest version of CaptureSDK.

cyrille-socket, 2024-11-18:
Hi @huynhlqDev, do you have the stack trace of the BLE crash please?