SocketMobile/swift-package-capturesdk #22: Issue: SDK Crashes on launch

adamjcampbell, 2025-03-28:
### Your Socket Mobile Developer ID

65003173-e722-e911-a968-000d3a36399f

### Environment

iOS 18
Capture SDK [1.9.110](https://github.com/SocketMobile/swift-package-capturesdk/releases/tag/1.9.110)
iPhone 

### Description

The app crashes on launch when using the CaptureSDK

### Steps to Reproduce

Run a SwiftUI app on the latest iOS with code similar to:
```
                    let helper = CaptureHelper.sharedInstance
                    helper.dispatchQueue = DispatchQueue.main
                    helper.pushDelegate(delegate)

                    let AppInfo = SKTAppInfo()
                    AppInfo.appID = "xxxx"
                    AppInfo.appKey = "xxxx"
                    AppInfo.developerID = "xxxx"
                    helper.openWithAppInfo(AppInfo) { result in
                        print("Capture result: \(result)")
                    }
```

In the App's `init` function

### Expected Behavior

The app doesn't crash

### Actual Behavior

The app crashes

### Additional Context

```
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFString containsObject:]: unrecognized selector sent to instance 0x303f5f9a0'
*** First throw call stack:
(0x18bea65fc 0x189421244 0x18bfff174 0x18bea95c8 0x18bea8ef0 0x1072281f4 0x107208b98 0x18aad8c9c 0x18aac8568 0x18aac8488 0x18aac68e0 0x18aafdce0 0x18aafd910 0x105bf38b0 0x105be2578 0x105be5680 0x105be496c 0x105bf6d34 0x105bf7724 0x216c34644 0x216c32474)
libc++abi: terminating due to uncaught exception of type NSException
```

```
Thread 4 Queue : CaptureServiceQueue (QOS: UNSPECIFIED) (concurrent)
#0	0x00000001dd5df2d4 in __pthread_kill ()
#1	0x0000000216c3959c in pthread_kill ()
#2	0x0000000193cf4b08 in abort ()
#3	0x0000000216b5f5b8 in abort_message ()
#4	0x0000000216b4dbac in demangling_terminate_handler ()
#5	0x00000001894232c4 in _objc_terminate ()
#6	0x0000000216b5e87c in std::__terminate ()
#7	0x0000000216b5e820 in std::terminate ()
#8	0x000000018942f9d8 in objc_terminate ()
#9	0x0000000105be258c in _dispatch_client_callout ()
#10	0x0000000105be5680 in _dispatch_continuation_pop ()
#11	0x0000000105be496c in _dispatch_async_redirect_invoke ()
#12	0x0000000105bf6d34 in _dispatch_root_queue_drain ()
#13	0x0000000105bf7724 in _dispatch_worker_thread2 ()
#14	0x0000000216c34644 in _pthread_wqthread ()
Exception Backtrace#0	0x000000018bea65fc in __exceptionPreprocess ()
#1	0x0000000189421244 in objc_exception_throw ()
#2	0x000000018bfff174 in -[NSObject(NSObject) doesNotRecognizeSelector:] ()
#3	0x000000018bea95c8 in ___forwarding___ ()
#4	0x000000018bea8ef0 in _CF_forwarding_prep_0 ()
#5	0x00000001072281f4 in -[SKTCaptureServiceDirect openClient:withAppInfo:completionHandler:] ()
#6	0x0000000107208b98 in __48-[SKTCapture openWithAppInfo:completionHandler:]_block_invoke ()
#7	0x000000018aad8c9c in __NSINDEXSET_IS_CALLING_OUT_TO_A_BOOL_BLOCK__ ()
#8	0x000000018aac8568 in -[NSBlockOperation main] ()
#9	0x000000018aac8488 in __NSOPERATION_IS_INVOKING_MAIN__ ()
#10	0x000000018aac68e0 in -[NSOperation start] ()
#11	0x000000018aafdce0 in __NSOPERATIONQUEUE_IS_STARTING_AN_OPERATION__ ()
#12	0x000000018aafd910 in __NSOQSchedule_f ()
#13	0x0000000105bf38b0 in _dispatch_block_async_invoke2 ()
#14	0x0000000105be2578 in _dispatch_client_callout ()
#15	0x0000000105be5680 in _dispatch_continuation_pop ()
#16	0x0000000105be496c in _dispatch_async_redirect_invoke ()
#17	0x0000000105bf6d34 in _dispatch_root_queue_drain ()
#18	0x0000000105bf7724 in _dispatch_worker_thread2 ()
#19	0x0000000216c34644 in _pthread_wqthread ()
#20	0x0000000216c32474 in start_wqthread ()
```

adamjcampbell, 2025-03-28:
Going back as far as version: `1.8.34` which is used by one of our other apps doesn't have this issue

cyrille-socket, 2025-03-28:
Hello @adamjcampbell,

usually this crash happens because you have forgotten or made a mistake with the keys to add into the `Info.plist` of your app when adding our CaptureSDK.

> For SocketCam C860 is an enhanced version of SocketCam C820, you also need to add the following key to your Info.plist: LSApplicationQueriesSchemes (Queried URL Schemes) with a new item: sktcompanion (in lower case).

Meaning that it's not a string but an array of strings as follows:

```
	<key>LSApplicationQueriesSchemes</key>
	<array>
		<string>sktcompanion</string>
	</array>

```

Please let me know if that fixes your issue.

Best,
Cyrille

cyrille-socket, 2025-03-31:
You should see the exception thrown in Debug mode and displays a message in your Xcode console:

> IN ORDER TO LAUNCH CAPTURESDK YOU NEED TO ADD INTO YOUR INFO.PLIST THE KEY 'LSApplicationQueriesSchemes' CONTAINING THE ELEMENT 'sktcompanion' ALL LOWERCASE. YOU CAN ALSO REFER TO THE DOCUMENTATION

In production, if it crashes, it means that it's not an array but I hope you test your app before releasing 😅

closed 2025-03-31 by cyrille-socket

adamjcampbell, 2025-03-31:
@cyrille-socket of course I test before release, not sure why that was a suggestion?

The nice debug message you quoted only appears if the element is completely missing. If it is misconfigured the SDK crashes with `unrecognized selector` in `SKTCaptureServiceDirect` in the stack trace shown above.

It was a suggestion that some type checking was added in the SDK when `LSApplicationQueriesSchemes` is pulled from the plist a better message could be thrown as opposed to the unrecognized selector crash.

Anyway, it's fixed now. Thanks for your help.