SocketMobile/swift-package-capturesdk #25: Issue: SwiftUI intergration

Angkur03, 2025-05-16:
### Your Socket Mobile Developer ID

a9781787-9562-4406-be7f-649994d77d8b

### Environment

ios 18

### Description

In swiftui project, I just get the device arrival notification. but after that it is not working. can some one provide an example of swiftui integration?

### Steps to Reproduce

just need the implementation in a swiftui project

### Expected Behavior

It should work like uikit based project

### Actual Behavior

its not worked!

cyrille-socket, 2025-05-16:
Hello @Angkur03,

the implementation in SwiftUI is really similar to UIKit.

What kind of scanner do you use? Bluetooth Classic or Bluetooth LE or Camera scanning?

Then I cannot give you a proper answer with just "it doesn't work". I need more information from you. Do you any logs or the behavior of the scanner, etc...

In the meantime, here's a quick implementation that retains a `selectedScanner` that you can apply get/set property through the CaptureHelper.

```
import CaptureSDK
import CaptureHelper
import SwiftUI

@MainActor
final class ScannerManager: ObservableObject {

    private let captureHelper = CaptureHelper.sharedInstance

    @Published var selectedScanner: CaptureHelperDevice? = nil

    init() {
        let captureHelper = CaptureHelper.sharedInstance
        captureHelper.dispatchQueue = DispatchQueue.main

        captureHelper.pushDelegate(self)

        let AppInfo = SKTAppInfo()
        AppInfo.appKey = ""
        AppInfo.appID = ""
        AppInfo.developerID = ""

        captureHelper.openWithAppInfo(AppInfo) { (result: SKTResult) in
            print("Result of Capture initialization: \(result.rawValue)")
        }
    }

}

extension ScannerManager: CaptureHelperErrorDelegate, CaptureHelperDevicePresenceDelegate, CaptureHelperDevicePowerDelegate, CaptureHelperDeviceDecodedDataDelegate, CaptureHelperDeviceManagerPresenceDelegate {
    // MARK: - CaptureHelperDeviceDecodedDataDelegate

    func didReceiveDecodedData(_ decodedData: SKTCaptureDecodedData?, fromDevice _: CaptureHelperDevice, withResult result: SKTResult) {
        guard let decoded = decodedData else {
            print("No decoded data received")
            return
        }

        guard result == .E_NOERROR else {
            print("No new barcode scanned, SKTCaptureError: \(result.rawValue)")
            return
        }

        guard let readData = decoded.decodedData else {
            print("Error getting decoded data")
            return
        }

        guard var readBarcode = String(data: readData, encoding: .utf8) else {
            print("Error creating barcode")
            return
        }

        // HANDLE YOUR DECODED DATA HERE
    }

    // MARK: - CaptureHelperDevicePresenceDelegate

    // Good place for:
    // - confirming to the user the scanner is present and ready
    // - switching to a view that is ready to receive the decoded data
    // - checking the scanner configuration before setting its configuration
    func didNotifyArrivalForDevice(_ device: CaptureHelperDevice, withResult _: SKTResult) {
        print("Device arrival: \(device.info.name)")

        selectedScanner = device
    }

    func didNotifyRemovalForDevice(_ device: CaptureHelperDevice, withResult _: SKTResult) {
        print("Device removal: \(device.info.name)")

        if selectedScanner == device {
            selectedScanner = nil
        }
    }

    // MARK: - DeviceManager Presence Delegate

    func didNotifyArrivalForDeviceManager(_: CaptureHelperDeviceManager, withResult _: SKTResult) {
        print("Device Manager arrival")
        
        // Set each device as the favorite to be able to reconnect each device again after turning off/on
        deviceManager.getFavoriteDevicesWithCompletionHandler { result, favorites in
            if result == SKTCaptureErrors.E_NOERROR, let favorites {
                // if favorites is empty (meaning NFC Reader/Writer auto-discovery is off)
                // then set it to "*" to connect to any NFC Reader/Writer in the vicinity
                // To turn off the BLE auto reconnection, set the favorites to
                // an empty string
                if favorites.isEmpty {
                    deviceManager.setFavoriteDevices("*") { result in
                        print("Setting new favorites returned \(result.rawValue)")
                    }
                }
            }
        }
    }

    func didNotifyRemovalForDeviceManager(_ device: CaptureHelperDeviceManager, withResult _: SKTResult) {
        print("Removal For Device Manager \(device.info.name ?? "")")
    }

    // MARK: - CaptureHelperErrorDelegate

    func didReceiveError(_ error: SKTResult) {
        print("Receive a Capture error: \(error.rawValue)")
    }

    // MARK: - CaptureHelperDevicePowerDelegate

    func didChangePowerState(_ powerState: SKTCapturePowerState, forDevice _: CaptureHelperDevice) {
        print("Power State \(powerState.rawValue)")
    }

    func didChangeBatteryLevel(_ batteryLevel: Int, forDevice _: CaptureHelperDevice) {
        print("Battery Level: \(batteryLevel)")
    }
}
```

Angkur03, 2025-05-18:
Hello thanks for your reply. 

I am using your code: 

```
import CaptureSDK
import CaptureSDK
import SwiftUI

@MainActor
final class ScannerManager: ObservableObject {

    private let captureHelper = CaptureHelper.sharedInstance

    @Published var selectedScanner: CaptureHelperDevice? = nil

    init() {
        let captureHelper = CaptureHelper.sharedInstance
        captureHelper.dispatchQueue = DispatchQueue.main

        captureHelper.pushDelegate(self)

        let appInfo = SKTAppInfo();
        appInfo.appKey="MC4CFQCxHUaZDcAdrJImPHL3uSXjO5SmfgIVAMnmAMnDyIhDFU9s9humm2nbpBDN";
        appInfo.appID="ios:com.nsales.nvisionmobileXII";
        appInfo.developerID="a9781787-9562-4406-be7f-649994d77d8b";

        captureHelper.openWithAppInfo(appInfo) { (result: SKTResult) in
            print("Result of Capture initialization: \(result.rawValue)")
        }
    }

}

extension ScannerManager: CaptureHelperErrorDelegate, CaptureHelperDevicePresenceDelegate, CaptureHelperDevicePowerDelegate, CaptureHelperDeviceDecodedDataDelegate, CaptureHelperDeviceManagerPresenceDelegate {
    // MARK: - CaptureHelperDeviceDecodedDataDelegate

    func didReceiveDecodedData(_ decodedData: SKTCaptureDecodedData?, fromDevice _: CaptureHelperDevice, withResult result: SKTResult) {
        guard let decoded = decodedData else {
            print("No decoded data received")
            return
        }

        guard result == .E_NOERROR else {
            print("No new barcode scanned, SKTCaptureError: \(result.rawValue)")
            return
        }

        guard let readData = decoded.decodedData else {
            print("Error getting decoded data")
            return
        }

        guard var readBarcode = String(data: readData, encoding: .utf8) else {
            print("Error creating barcode")
            return
        }

        // HANDLE YOUR DECODED DATA HERE
    }

    // MARK: - CaptureHelperDevicePresenceDelegate

    // Good place for:
    // - confirming to the user the scanner is present and ready
    // - switching to a view that is ready to receive the decoded data
    // - checking the scanner configuration before setting its configuration
    func didNotifyArrivalForDevice(_ device: CaptureHelperDevice, withResult _: SKTResult) {
        print("Device arrival: \(device.deviceInfo.name)")

        selectedScanner = device
    }

    func didNotifyRemovalForDevice(_ device: CaptureHelperDevice, withResult _: SKTResult) {
        print("Device removal: \(device.deviceInfo.name)")

        if selectedScanner == device {
            selectedScanner = nil
        }
    }

    // MARK: - DeviceManager Presence Delegate

    func didNotifyArrivalForDeviceManager(_: CaptureHelperDeviceManager, withResult _: SKTResult) {
        print("Device Manager arrival")
        
//        // Set each device as the favorite to be able to reconnect each device again after turning off/on
//        deviceManager.getFavoriteDevicesWithCompletionHandler { result, favorites in
//            if result == SKTCaptureErrors.E_NOERROR, let favorites {
//                // if favorites is empty (meaning NFC Reader/Writer auto-discovery is off)
//                // then set it to "*" to connect to any NFC Reader/Writer in the vicinity
//                // To turn off the BLE auto reconnection, set the favorites to
//                // an empty string
//                if favorites.isEmpty {
//                    deviceManager.setFavoriteDevices("*") { result in
//                        print("Setting new favorites returned \(result.rawValue)")
//                    }
//                }
//            }
//        }
    }

    func didNotifyRemovalForDeviceManager(_ device: CaptureHelperDeviceManager, withResult _: SKTResult) {
        print("Removal For Device Manager \(device.deviceInfo.name ?? "")")
    }

    // MARK: - CaptureHelperErrorDelegate

    func didReceiveError(_ error: SKTResult) {
        print("Receive a Capture error: \(error.rawValue)")
    }

    // MARK: - CaptureHelperDevicePowerDelegate

    func didChangePowerState(_ powerState: SKTCapturePowerState, forDevice _: CaptureHelperDevice) {
        print("Power State \(powerState.rawValue)")
    }

    func didChangeBatteryLevel(_ batteryLevel: Int, forDevice _: CaptureHelperDevice) {
        print("Battery Level: \(batteryLevel)")
    }
}

```

these are the logs: 

Device arrival: Optional("SocketCam C820")
Device Manager arrival
Result of Capture initialization: 0
Device arrival: Optional("SocketCam C820")
Device Manager arrival
Result of Capture initialization: 0
Device arrival: Optional("SocketCam C820")
Device Manager arrival
Result of Capture initialization: 0
Device arrival: Optional("SocketCam C820")
Device Manager arrival
Result of Capture initialization: 0
Device arrival: Optional("SocketCam C820")
Device Manager arrival
Result of Capture initialization: 0
Device arrival: Optional("SocketCam C820")
Device Manager arrival
Result of Capture initialization: 0

i am using it from a view. where i just get the arrival status. But turing off the device or scanning anything is not notified there.

struct OrderlineProductPickerView: View {
    @State var viewModel: OrderlineItemViewModel
    @State var searchText: String = ""
    @Binding var showSearchBar: Bool
    @State var isOrderlineAddItemViewPresented: Bool = false
    
    @AppStorage(AppConstant.userSelectedPreviousItemSortBy) var previousSelectedSortBy = AppConstant.defaultSortBy
    
    @State var scannerManager = ScannerManager()

cyrille-socket, 2025-05-19:
Hi @Angkur03,

What kind of scanner do you use? Bluetooth Classic or Bluetooth LE or Camera scanning?

Because SocketCam C820 is a "virtual" device for the camera scanning. So it's not removed if you turn off the Bluetooth.

Thanks,
Cyrille

Angkur03, 2025-05-19:
I am not also receiving the barcode scanning notification also.

cyrille-socket, 2025-05-19:
The scanner SocketScan S800 is a Bluetooth Classic scanner, you need to pair it first in the iOS Bluetooth settings. Then you'll get the device arrival in your app.

Angkur03, 2025-05-20:
why I am receiving the barcode scanning result? 
And if i disconnected with from device bluetooth. it is not disconnected from the app. the published event are not notified.

cyrille-socket, 2025-05-20:
Can you send me a Github repository with your code that I can clone to try on my end please?

Angkur03, 2025-05-20:
```
import CaptureSDK
import CaptureSDK
import SwiftUI
import CaptureSDKFramework

@Observable
public class ScannerLogicSwift: NSObject {
    public static let sharedInstance = ScannerLogicSwift()
    var device : CaptureHelperDevice?
    var captureHelper = CaptureHelper.sharedInstance
    
    var overlay : UIViewController?
    
    override init() {
        super.init()
        
        self.captureHelper.pushDelegate(self)
        self.captureHelper.dispatchQueue = DispatchQueue.main
        let appInfo = SKTAppInfo();
        appInfo.appKey="MC4CFQCxHUaZDcAdrJImPHL3uSXjO5SmfgIVAMnmAMnDyIhDFU9s9humm2nbpBDN";
        appInfo.appID="ios:com.nsales.nvisionmobileXII";
        appInfo.developerID="a9781787-9562-4406-be7f-649994d77d8b";
        
        self.captureHelper.openWithAppInfo(appInfo, withCompletionHandler: { (_ result: SKTResult) in
            print("Socket: Result of Capture initialization: \(result.rawValue)")
            self.captureHelper.setConfirmationMode(.modeDevice, withCompletionHandler: { (result) in
                print("Socket: Data Confirmation Mode returns : \(result.rawValue)")
            })
        })
        
    }
    
    public func isScannerConnected() -> Bool
    {
        print("scanner connectd: \(self.device != nil)")
        return self.device != nil
    }
    
    func connectedDeviceName() -> String?
    {
        return self.device?.deviceInfo.name
    }
    deinit {
        print("Socket: ScannerLogic deinited")
    }
    
    func enableSwiftSoftCamWithOverlay(_ overlay: UIViewController) {
        if let scanner = device  {
            scanner.setTrigger(.start) { result, _ in
                print("Socket: scanned triggered: \(result.rawValue)")
            }
        }else {
            self.captureHelper.pushDelegate(self)
            self.captureHelper.dispatchQueue = DispatchQueue.main
            let appInfo = SKTAppInfo();
            appInfo.appKey="MC4CFQCxHUaZDcAdrJImPHL3uSXjO5SmfgIVAMnmAMnDyIhDFU9s9humm2nbpBDN";
            appInfo.appID="ios:com.nsales.nvisionmobileXII";
            appInfo.developerID="a9781787-9562-4406-be7f-649994d77d8b";
            
            self.captureHelper.openWithAppInfo(appInfo, withCompletionHandler: { (_ result: SKTResult) in
                print("Socket: Result of Capture initialization: \(result.rawValue)")
                self.captureHelper.setConfirmationMode(.modeApp, withCompletionHandler: { (result) in
                    print("Socket: Data Confirmation Mode returns : \(result.rawValue)")
                })
            })
            self.captureHelper.setSocketCamStatus(.enable) { result in
                print("Socket: Data Confirmation Mode returns : \(result.rawValue)")
            }
        }
[... 394 lines cut ...]
            Text("Scan a barcode")
                .font(.headline)
            
            if !scannedCode.isEmpty {
                Text("Scanned: \(scannedCode)")
            }
            
            Button("Trigger Scan") {
                ScannerLogicSwift.sharedInstance.triggerScannerButton()
            }
        }
        .onScannerNotification { code in
            self.scannedCode = code
        }
        .onAppear {
            ScannerLogicSwift.sharedInstance.enableSwiftSoftCamWithOverlay(UIViewController())
        }
    }
}

```

this is my full code. pls just create a new app. 
in camera scanner it works fine. but from hardware scan it did not receive any thing.
pls create a new project with the initial view as ScannerDemoView

Angkur03, 2025-05-20:
sorry my bad! actually the issue is in info plist.

cyrille-socket, 2025-05-20:
I Indeed forgot to ask you this part thinking it was ok by default. The importance of fully reading our documentation 😉

Angkur03, 2025-05-20:
thanks for your help :D