SocketMobile/swift-package-capturesdk #26: Issue: Rumba crashes

akamakc, 2025-05-20:
### Your Socket Mobile Developer ID

-

### Environment

iPadOS version 18.5
Rumba version 1.0.131 (latest)

### Description

When I start the Rumba app, a message appears saying that the certificate is not secure. I click on “Trust” and the app crashes.

[image]

### Steps to Reproduce

Install the Rumba app on iPad
Start the App
Open a website using certificate
Click on "Trust"

### Expected Behavior

the certificate will be trusted

### Actual Behavior

the Rumba app crashes

cyrille-socket, 2025-05-20:
Hello @akamakc,

please provide the exact url. I cannot reproduce with common urls that I tried.

Also, can you provide the version of Rumba please.

Thank you
Cyrille

akamakc, 2025-05-21:
> Hello [@akamakc](https://github.com/akamakc),
> 
> please provide the exact url. I cannot reproduce with common urls that I tried.
> 
> Also, can you provide the version of Rumba please.
> 
> Thank you Cyrille

Hello Cyrille,

sorry, that was my mistake. Now everything works with the certificate :) I am going to close this thread.

Thanks!

closed 2025-05-21 by akamakc