SocketMobile/swift-package-capturesdk #27: Query: CaptureSDK Crypto-Swift dependency.

George-Smudge, 2025-05-22:
### Your Socket Mobile Developer ID

830661fa-154a-ef11-a317-0022480a839f

### Environment

Updating from CaptureSDK 1.7.9 to 1.9.133.

### Description

Hi there,

I am in the process updating an iOS application to the latest version of CaptureSDK.
We have noticed that after updating Xcode now shows a new 3rd party sub-dependency “Crypto-Swift”. Is this new? If so our cybersecurity team would like to know if you would be able to provide any security policies, or a rationale for going with “Crypto-Swift” over Apples built in cryptography APIs. This is so we can plan our risk mitigation strategy for our change.

Many thanks,
George

### Steps to Reproduce

N/A

### Expected Behavior

N/A

### Actual Behavior

N/A

cyrille-socket, 2025-05-22:
Hello @George-Smudge,

the [CryptoSwift library](https://github.com/krzyzanowskim/CryptoSwift) was before included directly into our CaptureSDK but some developers had issues when using it also in their apps because of duplicated symbols. So that's why it is now a dependency. See [this](https://github.com/SocketMobile/swift-package-capturesdk/issues/7).

So to answer, it is not new.

We use CryptoSwift because another library that we use is using it for our NFC readers when they are in Writer mode (feature still in development on our CaptureSDK).

If you need more info or more specific inputs, please let me know.

Thanks
Cyrille

cyrille-socket, 2025-07-29:
No activity or reply. This thread is closed for now.

closed 2025-07-29 by cyrille-socket