SocketMobile/swift-package-capturesdk #28: Issue: The signature of SwiftDecoder.xcframework cannot be verified

adamjcampbell, 2025-06-10:
### Your Socket Mobile Developer ID

65003173-e722-e911-a968-000d3a36399f

### Environment

Xcode: 16.4

### Description

App with the Capture SDK fails to build with the error: The signature of “SwiftDecoder.xcframework” cannot be validated and may have been compromised.

### Steps to Reproduce

Integrate the capture sdk into an app and try to build

Or run `codesign -vv` on the SwiftDecoder.xcframework dependency

E.g.
```
➜ codesign -vv SwiftDecoder.xcframework
SwiftDecoder.xcframework: CSSMERR_TP_CERT_REVOKED
```

### Expected Behavior

The app can build and that signatures haven't been revoked

### Actual Behavior

The app cannot build the signature of the dependency has been revoked

### Additional Context

This is preventing building new app versions with the socket SDK

cyrille-socket, 2025-06-10:
Hello @adamjcampbell,

I'll notify the author of this SDK and get back to you once I have news from them.

Thanks
Cyrille

adamjcampbell, 2025-06-24:
Can confirm the issue is resolved. Thanks @cyrille-socket

closed 2025-06-24 by adamjcampbell