SocketMobile/swift-package-capturesdk #30: Issue: Unable to read barcodes from screens

benpackard, 2025-07-28:
### Your Socket Mobile Developer ID

95d04909-54d4-ee11-9079-000d3a5cb302 

### Environment

iOS 18.5 
CaptureSDK 1.19.133 
D860

### Description

Our D860 scanners (running in auto/presentation mode) are able to read barcodes printed on paper and even barcodes on physical products. But they do not read barcodes displayed on screens. 

Is there anything in the SDK required to enable/improve this?

### Steps to Reproduce

Attempt to scan a barcode from a mobile screen using the D860

### Expected Behavior

Barcode detected. 

### Actual Behavior

No barcode is detected.

cyrille-socket, 2025-07-29:
Hello @benpackard,

normally it should work. We have a tests station connected to a scanner that scans barcodes on an iPad screen for instance.

I'll gather some info on that and get back to you shortly.

Cyrille

cyrille-socket, 2025-07-30:
Hello @benpackard,

so normally with an angle of 45 degrees with laser beam and the screen should work. Our Support team will contact you on this. Please give us your email address.

benpackard, 2025-07-30:
Thanks. Please have them contact me at ben.packard@spargoinc.com

benpackard, 2025-08-05:
Hi, unfortunately that is not the outcome. I’m not sure where that information came from.  That was recommended as a solution yesterday but is not tested yet on our end. It will take some time to set up our test suite and test with our IT team. 

I don’t mind if this ticket is closed while I continue working with your support folks instead, but the issue is not yet resolved.